grblHAL/SAM3X8E #2: Arduino DUE Core onboard EEPROM

qiwalker, 2021-08-03:
hi,

i found this Due Core baord which is have EEPROM

https://www.google.com/url?sa=i&url=http%3A%2F%2Fwww.inhaos.com%2Fproduct_info.php%3Fproducts_id%3D164&psig=AOvVaw1lks_i1abs69cxyKPKZWJq&ust=1628044008954000&source=images&cd=vfe&ved=0CAwQjhxqFwoTCMjMh9nmk_ICFQAAAAAdAAAAABAN

can it use with GRBL HAL ?

thanks

terjeio, 2021-08-03:
> can it use with GRBL HAL ?

The I2C code needs to be modified for that as it is connected to another port than the current code supports. You can do that?

qiwalker, 2021-08-04:
i can't, no basic for C++ :(

terjeio, 2021-08-04:
> no basic for C++

grblHAL is written in plain C...

Next build will have support for larger EEPROMS (> 2K) and the port used by this board. Add
`#define I2C_ENABLE 2`
to the board map file and set [EEPROM_ENABLE](https://github.com/grblHAL/SAM3X8E/blob/781d7b0a490553ea3e5b11acdfacfd024f169fc0/grblHAL_Due/src/my_machine.h#L41) to `2` in _my_machine.h_ when uncommenting.

I am not sure when this build will be committed - I have some other changes I have to complete first.

closed 2021-08-04 by terjeio