grblHAL/SAM3X8E #6: Inverting limit switches does not work.

farbefreak, 2022-03-09:
Hello Guys, 

i compiled grblHAL with your guide(Arduino) for a Arduino Due 
I added a my_machine_map.h to reflect the pinout used by the Arduino CNC shield. 
I tried to upgrade from a Arduino UNO to a Due to give more raster engraving performance.

Everything seems to work fine except for the Limit switches. 
I was running NO switches to GND (not best practice, I know), without problems with the Arduino UNO and grbl.
With grblHAL i cant get that to work. 
As grblHAL expects NC to GND, I would have needed to invert the limit switches. 

I expected: 
Without inverting, the limit switches should be trigged without the switches pressed as the pin would be high without the limit switch pressed

I got (without inverting the switches with $5=1):
Switches show trigged without pressing them. Pressing the switches shows "not trigged" -> everything seems correct.

However I got (WITH inverting the switches $5=1): 
Switches show trigged with and without pressing them. They show no reaction to the pin state. 
I checked, the pull-ups are still activated as I have 3,3V on the limit switch pins without pressing them (and 0V if I press them).

Seems like something messes up the inverted state of these pins?

terjeio, 2022-03-09:
> Seems like something messes up the inverted state of these pins?

Yep - bad code, change this:
https://github.com/grblHAL/SAM3X8E/blob/d63ab30f14b5778bae2589229bb4e10940f900e0/grblHAL_Due/src/driver.c#L602
to 
`    signals.min.mask = settings.limits.invert.mask;`

BTW `$5=1` only inverts the X input, to invert all set `$5=7`.

farbefreak, 2022-03-09:
Thanks for your quick response!

I will test that tomorrow and will update here.

closed 2022-04-16 by terjeio