grblHAL/SAM3X8E #7: Ganged axis always moves on same direction

Idire, 2022-04-07:
Hello everyone,

I've been trying to get grblHAL to work on a g2core shield that I have after not being completely happy with that firmware, but I'm running into some problems that I can't debug.

The main issue is that my machine has two motors for the Y axis, so I've enabled `Y_GANGED` and `Y_AUTO_SQUARE`, but the secondary Y axis moves only on one direction all the time, e.g. if I ask for a positive movement primary Y moves positive and so does secondary Y, but if I ask for a negative movement primary Y moves negative but secondary Y moves positive again. I've swapped motors, drivers, wirings and pins assigned to secondary motor on `my_machine_map.h` and the problem remains, so I can't think about any other source but software.

Thank you in advance for any help.

terjeio, 2022-04-10:
It is due to a typo in this line:

https://github.com/grblHAL/SAM3X8E/blob/cb0f29df63469ce3cc3d4ac3f65846e2a1d80dc4/grblHAL_Due/src/driver.c#L194

It should be written like this:

`#ifdef Y2_DIRECTION_PIN`

Can you try changing it and report back?

terjeio, 2022-04-16:
Fix committed.

closed 2022-04-16 by terjeio