grblHAL/SAM3X8E #10: Keypad

schlindw, 2022-07-20:
Hello,

I tried to enable the second serial port to use it for a keypad/MPG. Unfortunately, the controller does not response to any command I am sending to it. I tried also the other ports than the defined one.

I started with the obvious modification of the my.machine.h:

```
#define KEYPAD_ENABLE 2
#define SERIAL2_DEVICE 1
```

After some research I added also this line, which is not original part of the file:
`#define MPG_ENABLE 1 (or 2)`

After some more research, I found out that other implementations include this part in their their driver.c, which is missing in the implementation for the Due:

```
    serialRegisterStreams();

#if MPG_MODE == 1
    if((hal.driver_cap.mpg_mode = stream_mpg_register(serialInit(115200), false, NULL)))
        protocol_enqueue_rt_command(mpg_enable);
#elif MPG_MODE == 2
    hal.driver_cap.mpg_mode = stream_mpg_register(serialInit(115200), false, keypad_enqueue_keycode);
#elif KEYPAD_ENABLE == 2
    stream_open_instance(0, 115200, keypad_enqueue_keycode);
#endif

```
After adding this section to my driver.c file, the compiling of that code fails because the 'keypad_enqueue_keycode' is unknown.

Anyone an idea what to do next?

Best regards
Georg

terjeio, 2022-07-20:
Add to the top of driver.c:
```
#if KEYPAD_ENABLE == 2
#include "keypad/keypad.h"
#endif
```
And try this code instead, it switches to the secondary UART stream (if not overridden) if USB_SERIAL_CDC is not enabled:
```
    serialRegisterStreams();

#if MPG_MODE == 1
  #if KEYPAD_ENABLE == 2
    if((hal.driver_cap.mpg_mode = stream_mpg_register(stream_open_instance(MPG_STREAM, 115200, NULL), false, keypad_enqueue_keycode)))
        protocol_enqueue_rt_command(mpg_enable);
  #else
    if((hal.driver_cap.mpg_mode = stream_mpg_register(stream_open_instance(MPG_STREAM, 115200, NULL), false, NULL)))
        protocol_enqueue_rt_command(mpg_enable);
  #endif
#elif MPG_MODE == 2
    hal.driver_cap.mpg_mode = stream_mpg_register(stream_open_instance(MPG_STREAM, 115200, NULL), false, keypad_enqueue_keycode);
#elif KEYPAD_ENABLE == 2
    stream_open_instance(KEYPAD_STREAM, 115200, keypad_enqueue_keycode);
#endif
```

schlindw, 2022-07-25:
Hello Terje Io,

I have seen that you have updated the code on github so I discarded my old one and started new from scratch. I found a minor bug in driver.c, there are two for-Loops where the i-variable is not declared as int. I think it was also neccessary to define the MPG_MODE, as I could not find this somewhere in the files.

So far, I can see on the second UART the feedback  that the controller sends via the primary connection if a grbl sender (PC) is running. The controller responds also to the realtime commands for jogging and simple G-code. However commands like '?', '^X', $xy=z", etc. seem to be ignored at the moment. 

What I would like to build is an external keypad like you and some other people have done it. What I am not sure if those can be operated completly standalone from a sender program on the computer? 
Often I need to start the PC and run a program just to put my milling machine a few centimeters out of the way, or drilling some holes with a certain space in between. I hoped to find with the keypad an easier way to do that, and in fact thats what I will realize. But for that solution I require some of the afforementioned commands, unlock, soft-reset, status query.

Best regards
Georg

terjeio, 2022-07-25:
> I found a minor bug in driver.c, there are two for-Loops where the i-variable is not declared as int. 

Thanks, I'll fix that.

>

terjeio, 2022-07-25:
>  I think it was also neccessary to define the MPG_MODE, as I could not find this somewhere in the files.

Correct, I did not add config for MPG mode as I have not tested the code yet. Replace
https://github.com/grblHAL/SAM3X8E/blob/ec25036de2cfc45a5c8749694f36fc3aef2e1f44/grblHAL_Due/src/my_machine.h#L33-L45
with

 ```
// Configuration
// Uncomment to enable.

//#define USB_SERIAL_CDC     1 // Use native USB port for communication.
//#define SAFETY_DOOR_ENABLE 1 // Enable safety door input.
//#define BLUETOOTH_ENABLE   1 // Set to 1 for HC-05 module. Requires and claims one auxillary input pin.
//#define VFD_ENABLE         1 // Set to 1 or 2 for Huanyang VFD spindle. More here https://github.com/grblHAL/Plugins_spindle
//#define MODBUS_ENABLE      1 // Set to 1 for auto direction, 2 for direction signal on auxillary output pin.
//#define PLASMA_ENABLE      1 // Plasma/THC plugin.
//#define OPENPNP_ENABLE     1 // OpenPNP plugin. To be completed.
#define MPG_ENABLE         1 // Enable MPG interface. Requires serial port and one handshake pin unless
                               // KEYPAD_ENABLE is set to 2 when mode switching is done by the CMD_MPG_MODE_TOGGLE (0x8B)
                               // command character. Set both MPG_ENABLE and KEYPAD_ENABLE to 2 to use a handshake pin anyway.
#define KEYPAD_ENABLE      2 // Set to 1 for I2C keypad, 2 for other input such as serial data. If KEYPAD_ENABLE is set to 2
                               // and MPG_ENABLE is uncommented then a serial stream is shared with the MPG.
//#define EEPROM_ENABLE      1 // I2C EEPROM support. Set to 1 for 24LC16 (2K), 3 for 24C32 (4K - 32 byte page) and 2 for other sizes. Uses eeprom plugin.
//#define EEPROM_IS_FRAM     1 // Uncomment when EEPROM is enabled and chip is FRAM, this to remove write delay.

```
> What I would like to build is an external keypad like you and some other people have done it. What I am not sure if those can be operated completly standalone from a sender program on the computer?

They can, there are two MPG modes - one where [switching is controlled by an input pin](https://github.com/terjeio/GRBL_MPG_DRO_BoosterPack#how-it-works) and one where switching is done by a real time command. The simplest is to use the real time command 0x8B to take control, when granted you will receive a real time report with the element `|MPG:1` present.

gschlin, 2022-07-30:
The problem remains the same, the second UART is blind for commands that are send via it (e.g. '?').
Therefore, the second UART displays the status messages, requested by the primary UART. And it processes Jog-commands.
But if I send 0x8B via the primary, the response is 'OK', but the mode is still |MPG:0 .
If I send 0x8B via the secondary, there is even no response at both UARTS and I have to press the reset button!

terjeio, 2022-07-31:
The serial code did not implement a required function, I have pushed a fix for that. Sorry about that.

closed 2022-09-27 by terjeio