grblHAL/SAM3X8E #11: Compile error in Arduino IDE 1.8.19

ozzyman, 2022-07-30:
I have to confess that I am a complete bumblefuck when it comes to C code, but I can't get the SAM3X38 driver to compile. It fails with the following error messages:

/home/tony/snap/arduino/current/Arduino/libraries/grblHAL_Due/src/driver.c: In function 'driver_init':
/home/tony/snap/arduino/current/Arduino/libraries/grblHAL_Due/src/driver.c:1663:9: error: 'i' undeclared (first use in this function)
     for(i = 0 ; i < sizeof(inputpin) / sizeof(input_signal_t); i++) {
         ^
/home/tony/snap/arduino/current/Arduino/libraries/grblHAL_Due/src/driver.c:1663:9: note: each undeclared identifier is reported only once for each function it appears in
Multiple libraries were found for "grblHAL_Due.h"
 Used: /home/tony/snap/arduino/current/Arduino/libraries/grblHAL_Due
 Not used: /home/tony/snap/arduino/current/Arduino/libraries/src
exit status 1
Error compiling for board Arduino Due (Programming Port).

There are some oddities:
The first is that the code snippet in the error does not appear at line 1663, but does appear in several other places. It also looks like the index variable _i_ IS declared as uint32_t a few lines above.

I have tried removing and re-adding the libraries, and compiing for the native USB port. Same problem.

ozzyman, 2022-07-30:
OK, I have surprised myself, and located the problem. the driver_init function does not declare i if the   NPG mode pin is not defined.

By changing: 

#ifdef MPG_MODE_PIN
    // Pull down MPG mode pin until startup is completed.
    uint32_t i = 0;
    while(mpg_pin == NULL) {
        if(inputpin[i].id == Input_ModeSelect) {
            mpg_pin = &inputpin[i];
            mpg_pin->bit = 1U << mpg_pin->pin;
            PIO_Mode(mpg_pin->port, mpg_pin->bit, OUTPUT);
            BITBAND_PERI(MPG_MODE_PORT->PIO_ODSR, MPG_MODE_PIN) = 0;
        }
        i++;
    }
#endif

to:

#ifdef MPG_MODE_PIN
    // Pull down MPG mode pin until startup is completed.
    while(mpg_pin == NULL) {
        if(inputpin[i].id == Input_ModeSelect) {
            mpg_pin = &inputpin[i];
            mpg_pin->bit = 1U << mpg_pin->pin;
            PIO_Mode(mpg_pin->port, mpg_pin->bit, OUTPUT);
            BITBAND_PERI(MPG_MODE_PORT->PIO_ODSR, MPG_MODE_PIN) = 0;
        }
        i++;
    }
#endif
    uint32_t i = 0;

I get it to compile.

terjeio, 2022-07-30:
I fixed this a few days ago but forgot to commit it as I have been busy with other issues. Will commit the fix later today.

ozzyman, 2022-08-01:
Thanks, Terje. 

I'm working up a machine map for RAMPS1.4 as well.

closed 2022-08-24 by ozzyman