grblHAL/SAM3X8E #13: Board disconnecting when ending jobs

dmtuniverse, 2022-09-09:
Thanks for the help getting up and running. Unfortunately I have now an issue with running jobs successfully.

Every time I manually end a job, or the job finishes, the board crashes.

Using IOsender I sometimes receive the error message "unhandled exception - COM port does not exist" when I manually stop a job, but more often than not the software just hangs and I can't kill the process in task manager until I physically unplug the USB cable.

Whenever a job finishes the result is the same, returning to home from the final coordinates causes the board to disconnect and hangs IOsender. I tried UGS and the same thing happened. Sometimes it happens in the middle of a job too, though not as often.

I have also tried running a job on the board with the board by itself (not in the machine, nothing connected except USB cable), and the same thing happens.

I can't monitor the console output at the time of it happening as it hangs, so I'm not even sure what line of gcode is causing this.

I love the speed grblHAL has allowed me to reach with laser engravings, but I am experiencing far too many crashes for it to be usable.

Have I done something wrong, missed a setting? Any log files I can check?

Thanks.

dmtuniverse, 2022-09-09:
`; LightBurn 1.0.06
; GRBL device profile, absolute coords
; Bounds: X11 Y13 to X41 Y31
G00 G17 G40 G21 G54
G90
M4
; Cut @ 100 mm/sec, 10% power
M8
G0X11Y13
M3
; Layer C00
G1Y31S100F6000
G1X41
G1Y13
G1X11
M9
G1S0
M5
G90
; return to user-defined finish pos
G0 X0 Y0
M2
`

Gcode example from LB which is what I am using to compile.

terjeio, 2022-09-09:
Your test program runs fine for me, and I can _Feed Hold_, _Stop_ or _Reset_ at any time (with program running or not) in ioSender with no issues.

> Using IOsender I sometimes receive the error message "unhandled exception - COM port does not exist" when I manually stop a job, but more often than not the software just hangs and I can't kill the process in task manager until I physically unplug the USB cable.

This means the controller crashes or has been hard reset. Note that a both a crash or hard reset will kill the USB stack so the reset button cannot be used when communicating via native USB.

I have an [Arduiono provided USB driver](https://www.arduino.cc/en/Guide/ArduinoDue) installed:

[image]

This is used for both the programming and native USB port. Do you have the same?

Random crashes may be due to electrical interference, or perhaps a bad USB cable/hub. It is a bit odd that you get consistent crashes on program end though. Is the machine a CO2 laser?

If you can get your computer to recognize the programming port then you may try that.

closed 2023-08-26 by terjeio