grblHAL/SAM3X8E #16: compiling error

Ramr93, 2023-01-01:
hi I'm trying to compile using Arduino IDE .  I add all the libraries . but it gives tons on error . please help me to compile with steps

terjeio, 2023-01-02:
See the [Wiki](https://github.com/grblHAL/core/wiki/Compiling-grblHAL).

Ramr93, 2023-01-03:
i tried BOSSA 1.9.1 also but i didn't work. the port is connected but it shows following error 
[image]

terjeio, 2023-01-03:
Did you follow the ["How to use"](https://mdac.com.au/2018/01/14/programming-the-arduino-due-with-bossa/) instructions? If this failed is your board broken? Time to contact the vendor?

terjeio, 2023-01-04:
>  can you provide any video tutorial to upload the grblhal on Arduino mega 2560?

No, since grblHAL cannot run efficiently (or at all?) on a 8-bit processor. So there is currently no 8-bit driver available and I have no plans to write any.

closed 2023-01-05 by Ramr93