grblHAL/SAM3X8E #18: Arduino Due flashed as ramps 1.6 homing fail

Kickpunchy, 2023-08-25:
My NC Limit switches trigger alarm when machine is jogged to limit. However, homing cycle crashes through z max limit. Tried with UGS and gsender.

terjeio, 2023-08-25:
Switches are wired correctly? Trigger each separately and check with $LIM command.

Kickpunchy, 2023-08-26:
yes, switches are wired correctly.

terjeio, 2023-08-26:
Homing works for me in my test setup. How are the switched wired (to min or max inputs or both)?

Kickpunchy, 2023-08-26:
Both

terjeio, 2023-08-26:
Are you homing towards max?

Kickpunchy, 2023-08-26:
Yes

terjeio, 2023-08-26:
Ok, that explains it. You have to home towards min.

Kickpunchy, 2023-08-26:
Really? OMG I did not even think of trying that(neither did anybody on Reddit)

Kickpunchy, 2023-08-26:
I was kinda hoping and dreading I was just a dumbdumb

Kickpunchy, 2023-08-26:
So, just home all axises towards min?

terjeio, 2023-08-26:
Yes.

Kickpunchy, 2023-08-26:
Thanks I appreciate it

breiler, 2023-08-27:
It was a long time ago that I tried grblHAL and I would like to understand this for the next person asking.

In the classic gnea/grbl there was only one input for each axis. It would not matter where you put your limit switches, as long as you configured the homing direction correctly. Default in GRBL $23=0 means that all limit switches needs to be placed in the positive direction for each axis.

And this has changed in grblHAL?

I am assuming that this is @Kickpunchy:s original post on reddit with the configuration that was used:
https://www.reddit.com/r/hobbycnc/comments/15ycwoc/homing_cycle_crashes_through_z_max_limit_switch/

terjeio, 2023-08-27:
> In the classic gnea/grbl there was only one input for each axis.

As you state legacy Grbl has no notion of min and max limit switches, grblHAL has, the core can even differentiate between limit switches and home switches. grblHAL [uses the min limit inputs for homing](https://github.com/grblHAL/core/blob/1d7872fc5cd08da3b79e252ddb0fa0e5986829f7/machine_limits.c#L63-L85) if the driver [does not support separate inputs](https://github.com/grblHAL/core/blob/1d7872fc5cd08da3b79e252ddb0fa0e5986829f7/grbllib.c#L261-L263) for homing. No driver currently support separate homing inputs but that may change in the future.

> It would not matter where you put your limit switches, as long as you configured the homing direction correctly.

This still applies, as long as the switches in the homing direction is connected to the min inputs. I guess I selected the min inputs since homing by default sets the home position in negative space.

> And this has changed in grblHAL?

I believe not. Note that very few driver/board combinations has both min and max limit inputs - this is new in the Grbl world and may cause confusion?

breiler, 2023-08-27:
> This still applies, as long as the switches in the homing direction is connected to the min inputs. I guess I selected the min inputs since homing by default sets the home position in negative space.

If the configuration says that homing should be done on the positive end of each axis ($23=0) but you'd need to wire the switches to the min-inputs, which normally should be put on the negative ends of each axis I'd say that it is a bit confusing.

Is it ok if I make a note of it on the wiki here on $21 that the min inputs should be used? 
https://github.com/grblHAL/core/wiki/Additional-or-extended-settings

terjeio, 2023-08-27:
> Is it ok if I make a note of it on the wiki here on $21 that the min inputs should be used?

Let me think about this a bit. Currently only the Arduino Due and the LPC176x drivers has max inputs available at the driver level so I could change the behaviour to remove confusion without affecting too many users. In addition only the limit switches used for homing should have interrupt disabled during homing if hard limits is enabled.

terjeio, 2023-09-03:
I have now committed an update that significantly changes how limit switches are used for homing.
If both min and max are available the switch used for homing will be in the configured direction, depending on the driver the opposite switch will still have hard limits enabled if configured. Some drivers will also keep hard limits enabled for switches belonging to axes not involved in the executing homing cycle. It is my intention to update all drivers to support this.
A new command, `$HSS`, will output the current switch status in a `[HOMES:<min>,<max>]` report. This can be used to check if wiring is correct. E.g. if the X min switch and Y max switch is triggered the output will be `[HOMES:X,Y]`.

@Kickpunchy The Due driver should be fully compliant with this change. If you have time can you please verify?

@breiler I think this change warrant a new Wiki page dedicated to homing. What do you think?

breiler, 2023-09-04:
I think that a page specifically for homing is a good idea. =)