grblHAL/SAM3X8E #19: arduino due not working

enjoylife1024, 2023-12-12:
I am not able to compile this to my arduino.  I had someone at my work help me and it says it complied and is on the due, except I can't get it to link to lightburn or lasergrbl.  It has the com port and I can read it through get board info in arduino ide.  That is about it.  I am getting this on the log file from lightburn.
[LightBurnLog.txt](https://github.com/grblHAL/SAM3X8E/files/13642507/LightBurnLog.txt)

Any help is much appreciated.  I am currently using a mega on my machine now but want to go to the 32 bit.

terjeio, 2023-12-12:
I am not familiar with the Lightburn log but could it be that "state:  1" is from the controller beeing in an alam state?
If so see this [wiki page](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings) on how to get rid of that.

enjoylife1024, 2023-12-12:
can I get you a log from laser grbl?  Is there a software I can download that would help you know what the issue is?  I had a real hard time getting it to compile and I know that is my issue on not knowing how to put them in directories.  I believe it has to  do with something in the com connection.  What driver should be in the com port section of device manager.  Right now mine says arduino due.  Is it supposed to be the sx...?

terjeio, 2023-12-12:
Are you running Windows? If so [ioSender](https://github.com/terjeio/ioSender/releases/tag/2.0.43) will tell you what is wrong and let you change settings - even if alarm mode is active.

If you have compiled a version that connects over the programming port then this is how the port enumerates on my Win10 machine:

[image]

I'll have to recompile with USB enabled to find out how that port enumerates.

enjoylife1024, 2023-12-12:
I am at work now.  I have built a 4x8 cnc that is running grbl on an arduino mega.  It runs good but I want the 32 bit processing speed.  Is it ok if I continue to ask questions when I get home to get the sorted out.  I do appreciate all your help.  This is amazing what you have made here.

enjoylife1024, 2023-12-12:
[image]
[image]
I get this in the serial monitor in arduino ide when I plug it in.  So the Com port is definitely active.

enjoylife1024, 2023-12-12:
When I unplugged and plugged back in with the serial monitor on it did this.
[image]

enjoylife1024, 2023-12-12:
ok I see that the baud rate was wrong and now I can get it to work in the serial monitor portion but only through programming port.  Is this the port I want?  Don't I want it plugged into the native port when running the cnc?  Do both ports not work together?  Again so sorry very new to the programming aspect I am an automotive engineer by trade.

terjeio, 2023-12-12:
From the output above I am pretty sure the controller is in alarm mode. Have you tried to either short the reset/estop input to ground or changing the $14 setting to invert the input as described in the [Wiki page](https://github.com/grblHAL/core/wiki/First-Run-Grbl-Settings)?

The programming port is ok to use for communicating with the controller. The USB port can be enabled at compile time but both cannot be used at the same time. The USB port has faster transfer rate that may improve high resolution engraving speed - but you should explore other settings first. These are max travel speed, acceleration and planner buffer size.

enjoylife1024, 2023-12-12:
I thank you soooo much for your help.  It just started working.  I am doing the estop thing now.  

I understand you can not use both ports at same time.  I was moving the cord from one to the other and only the programming port was working.  You say the usb port can be enabled at compile time?  I guess I don't know what that means.  Would I have to change one of the files in your program or is it a variable that I can type in gcode screen.  Forgive my ignorance. 

Now that I can at least use it I can figure out what needs to change.  You mention max speed, acceleration and buffersize.  I want to travel at 30,000 mm/mm max.  That is what the water jets work at.  My cnc is an all in one.  I am figuring out all the logistics as I go.

terjeio, 2023-12-13:
>  You say the usb port can be enabled at compile time? 

The easiest way to compile for the Due is to use the [Web Builder](http://svn.io-engineering.com:8080/?driver=SAM3X8E). The _Connection_ dropdown can be used to select which port to use for communication, _(Virtual) UART_ selects the programming port, _Native USB_ the other one. If compiling locally uncomment [this line](https://github.com/grblHAL/SAM3X8E/blob/678b499cf3febed90a2762b8e734e49c1d9bf2db/grblHAL_Due/src/my_machine.h#L37) for native USB.