grblHAL/SAM3X8E #20: Arduino DUE and Ramps 1.6 - Can't find the pins for the second Y-axis

Helle62, 2024-01-06:
**Happy new year!**

I have two systems build for my CNC maschine to test which is better.
The first system is with a BlackPill and CNC-shield. 4 axis with ganged Y-axis. This system runs well.

Then I have build a second system with Arduino DUE and Ramps 1.6, because Ramps have 5 axis and 2 hard limits pro axis.
But the second/ganged Y-axis does not work. I have testet the second axis on the Ramps board on E0 and E1, but without success.

Have you an idea where are my mistake? Attached my json-file to see my configuration what I have build.
[SAM3X8E_RAMPS_16.json](https://github.com/grblHAL/SAM3X8E/files/13849449/SAM3X8E_RAMPS_16.json)

Many thanks in advanced,
Helmut

terjeio, 2024-01-06:
A Happy New Year for you as well.

There is a typo for the E1 enable pin in the map file. Can you try a new Web Builder build and report back if it works?

Helle62, 2024-01-06:
**Yeah great, it works now!**
Many thanks and have a nice rest of the weekend!

closed 2024-01-06 by Helle62