grblHAL/SAM3X8E #21: laser always on

enjoylife1024, 2024-02-01:
So with the arduino due my laser is always on.   I have a mega also, lightburn had an update so I used the two to make sure it was only the due.  When I plug it in the mega the laser acts like it should.  However when I plug in the due the laser is on all the time.  When i fire the laser my fan on my atomstack laser turns off.  I tried to see the difference between the mega load I have for mapping and the due.  The only thing I see is on the due it has this, #define SPINDLE_PWM_PIN     23 // Due Digital Pin 7 / PWML6 B.  I do not know what the pwm6l b is.  The mega does not have that.  Any help would be appreciated.  

Possible solution Could I just take the mega map I have and open the due mega map delete it all and paste in the mega map I know works?

enjoylife1024, 2024-02-01:
I am using grblahl on my due. The pwml6 b is the line from the mega map in grblhal. I tried two that I have as they are just screw terminals. No ramps no anything. Just using the due with a screw terminal hat. It acts as if maybe my pin 7 is set at high instead of low. I don’t know if there is a $ variable that needs to be set or not.

It is the Due MCU PWM timer/perihpheral used used to generate the pin output. The Mega map references OCR4C which I assume is a register, this does not exist in the Due MCU.

The Mega map<https://github.com/gnea/grbl-Mega/blob/edge/grbl/cpu_map.h> from legacy Grbl? That cannot be used for the Due since the MCU register/pin designations are not the same.

When I test the driver PWM output is at pin 7 as it should, this is the same pin as used in the legacy Grbl map.
Which breakout board do you use?
And are you running legacy Grbl on the Mega board, not some other fork with a different map?

terjeio, 2024-02-01:
Pin 7 is a PWM output, not a simple on/off. The pulse width (and thus the measured voltage) is controlled by the S value.
S = 0 -> no output, S = $30 setting -> 100% output. Set $32=1 for laser mode. This will require cutting motion (G1/G2/G3) to turn the laser on.
Do you have a multimeter and thus can measure the output?

enjoylife1024, 2024-02-01:
yes I do I can try all of this and send you results.  For my software lightburn the s value is set for the max rpm.  Which I have set at 10000.  Maybe I am just not understanding something as I am not a coder just a guy that tinkers with stuff.  $32 is set to 1.  I can get a print out of my gcode settings if this would help.  I can do any and all test you would like me to do.  I even have an oscilloscope.

enjoylife1024, 2024-02-01:
my $30-32 are all the same from the mega to the due

enjoylife1024, 2024-02-01:
maybe I should just reflash it?  Maybe I didn't flash it correctly.

terjeio, 2024-02-01:
>  I even have an oscilloscope.

Ok, then set $32 to 0 and check the output on the pin after enabling output with M3 and then sending different S-values. E.g. S5000 should result in a 50/50 duty-cycle square wave (assuming $30 is set to 10000). FYI the default $33 setting for grblHAL is 5000 Hz, IIRC legacy Grbl uses 1000 Hz  - this might be an issue so best to leave it at 5000 Hz. Higher is often better when doing gray scale engraving since the laser power response time will typically decrease a bit.

terjeio, 2024-02-01:
> maybe I should just reflash it? Maybe I didn't flash it correctly.

Since you can communicate with the board it is likely ok. What you should do is to check the `$I` output to validate that the board map is for the Mega. You can also check the pin assignments with the `$pins` command.

enjoylife1024, 2024-02-01:
I am not at home yet but I can check all of this when I am.  I think it my be an issue with lightburn but not sure. I will start debugging around 5:30 tonight.

enjoylife1024, 2024-02-01:
ok I did the pin out and it does not look like anything I understand sorry.  I also did voltage test and it is at 3.3v on pin 7 just plugged into the computer and 3.3v with lightburn on.

[PIN:A.0,Probe]
[PIN:B.17,Reset]
[PIN:B.18,Feed hold]
[PIN:B.19,Cycle start]
[PIN:C.29,X limit min]
[PIN:D.7,Y limit min]
[PIN:D.8,Z limit min]
[PIN:A.15,X step]
[PIN:D.0,Y step]
[PIN:D.1,Z step]
[PIN:D.9,X dir]
[PIN:A.7,Y dir]
[PIN:D.10,Z dir]
[PIN:B.27,X enable]
[PIN:C.24,Spindle on]
[PIN:C.25,Spindle direction]
[PIN:C.22,Flood]
[PIN:C.21,Mist]
[PIN:A.8,RX,Primary UART]
[PIN:A.9,TX,Primary UART]
[PIN:C.23,Spindle PWM]

well now looking at the code I understand that spindle pwm is the pin 7 but I have power to it.

enjoylife1024, 2024-02-01:
even with the $32 set to 0 it is still giving me a 3.3v on the pin 7

enjoylife1024, 2024-02-01:
I think I have it working now. I have to test in the laser tomorrow.  The $32 can't be 10000.  It did not give me a nice sign wave at all with it at that.  So I put it to 1000 and it looks a lot better.

terjeio, 2024-02-05:
> Nothing I tried worked. I am not sure why there is always power to this pin.

Perhaps a broken MCU then? I have managed to destroy a single pin a few times with different MCUs.

> I am thinking I do not have the right config for some reason. 

When I compile with the Mega map provided it works for me.

> I figured a lot out on the fly like having to adjust the file to only program the mega pin out instead of all of them. Is there a simple way for us tinker people to just pull the grblhal with the mega pin out only?

Not really, it migh be easier to just write a short .ino (sketch) to test the pin?

You may try to measure the voltage with the reset button pressed - if I am not mistaken this should put the pin in high Z mode and if you pull the pin to ground with a resistor, say 1K or so, it should measure close to or 0V.

closed 2024-02-22 by terjeio