grblHAL/SAM3X8E #24: Analog Aux Inputs on Due

JackDaviesDesign, 2024-05-20:
I am trying to add an analog input to measure the arc voltage of a plasma cutter. 

This is with no changes to the ramps_smart_map.h file.
`[PIN:A. 14,Aux in 0,Cutter up]
[PIN:D. 0,Aux in 1,Cutter down]
[PIN:D. 2,Aux in 2,Arc ok]`

I then added these lines to the ramps_smart_map.h file
`#define AUXINPUT3_PORT      PIOB
#define AUXINPUT3_PIN       20   // Analog input for THC - Due D65/A11  // AUX-2 Connector
`

This is the result of `$pins`
`[PIN:A. 14,Aux in 0,P0]
[PIN:D. 0,Aux in 1,Cutter up]
[PIN:D. 2,Aux in 2,Cutter down]
[PIN:B. 20,Aux in 3,Arc ok]`

Everything got shifted down a level. Do I need to define that this pin (B20) is an analog pin somewhere?

terjeio, 2024-05-20:
I have not added analog input capability to this driver yet so it is not possible to use voltage driven THC until that is done.

How do plan to connect arc voltage? More or less directly to an analog capable MCU pin? Or via galvanic isolation?

FYI Phil Barrett has made an [I2C ADC board](https://github.com/grblHAL/Plugin_plasma/discussions/6) that provides isolation, aimed at plasma use. Adding support for that may be a tad easier (and safer) than using a MCU pin.

Anyway code has to be added and I am not able to do that for a while since I am away and without access to a dev board. If you are capable of adding it then the driver that might have the closest matching bits could be the STM32Fxx driver, the most complex part is in [this file](https://github.com/grblHAL/STM32F7xx/blob/main/Src/ioports_analog.c).

JackDaviesDesign, 2024-06-11:
Thank you for the feedback, that would explain it!

I have since purchased an ADC board from Phil based on your recommendation. Would be great if you could look at implementing this for the SMART Ramps when you are back/have time. I looked at your examples but it goes a bit over my head!

Thank you in advance!

terjeio, 2024-06-20:
MCP3221 support added in the latest commit - but not yet available from the Web Builder.

terjeio, 2024-06-25:
MCP3221 support now available in the Web Builder, enable in the _Optional inputs_ tab.

closed 2024-08-23 by JackDaviesDesign