grblHAL/SAM3X8E #28: Spindle is switched off when speed is 0

Deltaplc, 2024-10-31:
Good morning
Sorry for my english....google translator....
I wanted to replace an Arduino Mega (grbl) with an Arduino Due SAM3X8E (grblhal) to control a laser and have more computing power!
I noticed that the enable pin doesn't work like in grbl! M3 S0 = enable pin off!! But it remains active even with just M3!!
Can its behavior be modified so that it is compatible with the behavior of the old grbl? I'm sorry to bother you but I can't find any documentation about it!!
Thank you in advance!

terjeio, 2024-10-31:
What is your $9 setting?

`$$=9` outputs:
```
    0 - Enable (1)
    1 - RPM controls spindle enable signal (2)
    2 - Disable laser mode capability (4)
Enable controls PWM output availability.
When `RPM controls spindle enable signal` is checked and M3 or M4 is active S0 switches it off and S > 0 switches it on.
```

And is $32=1 (laser mode)?

Deltaplc, 2024-10-31:
HI
Thanks for the reply!
The current position of $9=0!
I tried changing to $9=1....but it doesn't change the behavior!!
$32=1 ok!

Deltaplc, 2024-10-31:
I read somewhere that the $7 command also exists!!
What is this for?
It doesn't appear in my $$ list!
Could you tell me where I can find updated documentation on grblhal's $ commands?

Deltaplc, 2024-10-31:
This is in my config.h file....

// Spindle settings (Group_Spindle)

/*! @name $9 - Setting_SpindlePWMOptions
*/
///@{
#if !defined DEFAULT_SPINDLE_ENABLE_OFF_WITH_ZERO_SPEED || defined __DOXYGEN__
#define DEFAULT_SPINDLE_ENABLE_OFF_WITH_ZERO_SPEED Off
#endif
///@}

Deltaplc, 2024-10-31:
I tried other settings...
$9 = 3
Seems to make the enable pin work like in the old grbl!
But there is a problem!! The PWM only works if $32 = 0!!
If $32=1 nothing works with M3!!

closed 2024-11-03 by Deltaplc