grblHAL/SAM3X8E #29: Ramps 1.6 enable pin 10

Deltaplc, 2024-11-03:
Good morning
I'm working with a ramps 1.6...
I enabled "COOLANT_FLOOD" on pin 8 and "COOLANT_MIST" on pin 9.
Is it possible to also activate pin 10 in parallel with pin 9, to also activate the third mosfet of the ramps?
Thank you

terjeio, 2024-11-07:
> Is it possible to also activate pin 10 in parallel with pin 9, to also activate the third mosfet of the ramps?

Soon, by defining pin 10 as an aux output and using the [eventout plugin](https://github.com/grblHAL/Plugins_misc/blob/main/eventout.c) to bind it to the mist on/off handler. Or if you do not want to wait for me to add it officially [add the plugin manually](https://github.com/grblHAL/Plugins_misc/tree/main?tab=readme-ov-file#bind-events-to-aux-output-ports) and recompile with it enabled.

Deltaplc, 2024-11-07:
> > Ho provato ad aggiungere il plugin ma ottengo questi errori!! Temo di non aver capito come fare....
> 
> Il tuo core (cartella grbl) è troppo vecchio.

I downloaded grblhal 15 days ago!!

Is this a change you recently made??

terjeio, 2024-11-07:
> C:\Users\Davide\Documents\Arduino\libraries\grblHAL_Due\src\Plugins_misc-main\eventout.c:201:14: error: 'Setting_ActionBase' undeclared (first use in this function)

It is declared [here](https://github.com/grblHAL/core/blob/f35d14a901db2e3aea7024d3a6d80269255dd61e/settings.h#L470).

> Is this a change you recently made??

Relatively recent, the references you get compilation errors for about 2 months ago. Are you sure you are compiling against a newer version? Check the `$I` output, the build date is the last element in the version string: ` [VER:1.1f.20241031:]`

Deltaplc, 2024-11-07:
> > C:\Users\Davide\Documents\Arduino\libraries\grblHAL_Due\src\Plugins_misc-main\eventout.c:201:14: errore: 'Setting_ActionBase' non dichiarato (primo utilizzo in questa funzione)
> 
> Viene dichiarato [qui](https://github.com/grblHAL/core/blob/f35d14a901db2e3aea7024d3a6d80269255dd61e/settings.h#L470) .
> 
> > Si tratta di una modifica che hai apportato di recente?
> 
> Relativamente recenti, i riferimenti per cui hai ricevuto errori di compilazione risalgono a circa 2 mesi fa. Sei sicuro di compilare con una versione più recente? Controlla l' `$I`output, la data di compilazione è l'ultimo elemento nella stringa di versione:` [VER:1.1f.20241031:]`

[VER:1.1f.20240801:]
[OPT:VNM0SL,100,1024,3,0]
Target buffer size found
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,SED]
[FIRMWARE:grblHAL]
[SIGNALS:RHSP]
[NVS STORAGE:*FLASH]
[DRIVER:SAM3X8E]
[DRIVER VERSION:240730]

This is the version I downloaded using GIT!!
How should I download the latest version!!??

Deltaplc, 2024-11-07:
> Viene dichiarato [qui](https://github.com/grblHAL/core/blob/f35d14a901db2e3aea7024d3a6d80269255dd61e/settings.h#L470) .
>
>
> Relativamente recenti, i riferimenti per cui hai ricevuto errori di [...]

I just re-downloaded the firmware for SAM3X8E.....I checked in the settings folder and the parameter is present!!
 In my version that I'm using, I checked and that parameter doesn't exist! As soon as I have some time I'll recompile it with the updated version and I'll let you know! 
Thank you very much for your help!

Deltaplc, 2024-11-07:
With the version downloaded today with git and with the plugin you provided me, it now compiles well!!
As soon as possible I will test it on ramps1.6!!
Thanks again for your help!
You are doing a fantastic job with grblhal!How can I offer you a coffee/beer??I would love to!

Deltaplc, 2024-11-13:
> Soon, by defining pin 10 as an aux output and using the [eventout plugin](https://github.com/grblHAL/Plugins_misc/blob/main/eventout.c) to bind it to the mist on/off handler. Or if you do not want [...]

HI 
Sorry to bother you again...
Finally I received another card... so I can do the tests on the bench, without risking destroying my laser!
I'm testing the plugin you recommended, but I can't activate the pin 10 output (PIOC 29)...
I'm afraid I don't understand how to do it!!

I defined the exit:
#define AUXOUTPUT1_PORT     PIOC 
#define AUXOUTPUT1_PIN      29   

Then you should activate it by combining it with 
$750 = ?  and $751 = ?

I would like it to be activated with the Mist enable (M7)

What am I doing wrong??
Could you give me an example?
Thank you in advance!

terjeio, 2024-11-13:
`$$=750` (and $751 etc.) lists the options:

```
$$=750
    0 - None
    1 - Spindle enable (M3/M4)
    2 - Laser enable (M3/M4)
    3 - Mist enable (M7)
    4 - Flood enable (M8)
    5 - Feed hold
Event triggering output port change.
NOTE: the port can still be controlled by M62-M65 commands even when bound to an event.
```
`$$=760` (and 761 etc.) lists the port syntax:
```
$$=760
Aux output port number to bind to the associated event trigger.
NOTE: A hard reset of the controller is required after changing this setting.
```
So setting `$750=3` and `$760=1` should let M7 control aux out 1.

Deltaplc, 2024-11-14:
> Quindi l'impostazione `$750=3`dovrebbe `$760=1`consentire a M7 di controllare l'uscita AUX 1.

Sorry to bother you again! 
But it still doesn't work....

I have defined all outputs as "AUXOUTPUT".
I tested them with the M62/M63 - M64/M65 command.....
They work well! 

So the `$750=3` setting should `$760=3` allow the M7 to control the AUX 3 output (Pin 10). 

Nothing happens!! The Aux3 output (pin10) DOES NOT activate!!
Where am I going wrong!?
Are there any other settings I should look at?

#define AUXOUTPUT0_PORT     PIOC
#define AUXOUTPUT0_PIN      14     // Pin 49
#define AUXOUTPUT1_PORT     PIOC
#define AUXOUTPUT1_PIN      13     // Pin 50
#define AUXOUTPUT2_PORT     PIOC
#define AUXOUTPUT2_PIN      12     // Pin 51

#define AUXOUTPUT3_PORT     PIOC
#define AUXOUTPUT3_PIN      29  // Due Digital Pin 10
#define AUXOUTPUT4_PORT     PIOD  
#define AUXOUTPUT4_PIN      7     // Due Digital Pin 11 
#define AUXOUTPUT5_PORT     PIOD
#define AUXOUTPUT5_PIN      8     // Due Digital Pin 12

terjeio, 2024-11-14:
What are the $pins and $i outputs, after hard resetting the controller?

Deltaplc, 2024-11-14:
> Quali sono gli output $pins e $i dopo aver eseguito un hard reset del controller?

Avvio streaming
$0=10.0
$1=25
$2=0
$3=0
$4=7
$5=0
$6=0
$9=3
$10=511
$11=0.010
$12=0.002
$13=0
$14=1
$15=0
$16=1
$17=0
$18=0
$19=0
$20=0
$21=0
$22=1
$23=3
$24=50.0
$25=500.0
$26=40
$27=3.000
$28=0.100
$29=0.0
$30=1000.000
$31=0.000
$32=1
$33=5000.0
$34=0.0
$35=0.0
$36=100.0
$37=0
$39=1
$40=0
$43=1
$44=1
$45=3
$46=0
$62=0
$63=3
$64=0
$65=0
$100=160.19995
$101=160.19995
$102=250.00000
$110=20000.000
$111=20000.000
$112=500.000
$120=1000.000
$121=1000.000
$122=10.000
$130=400.000
$131=300.000
$132=200.000
$341=0
$342=30.0
$343=25.0
$344=200.0
$345=200.0
$346=1
$370=0
$372=0
$384=0
$394=4.0
$398=100
$481=0
$486=0
$673=1.0
$750=3
$751=0
$752=1
$753=2
$760=3
$761=0
$762=1
$763=2

Deltaplc, 2024-11-14:
In theory other outputs should also be activated.... but nothing is activated....

terjeio, 2024-11-14:
You posted the $$ output, I want to see $pins and $i...

Deltaplc, 2024-11-14:
Sorry I misunderstood!!

> You posted the $$ output, I want to see $pins and $i...

 $pins
[PIN:A.22,Reset]
[PIN:A.4,Feed hold]
[PIN:A.6,Cycle start]
[PIN:B.13,Probe]
[PIN:C.28,X limit min]
[PIN:B.25,X limit max]
[PIN:D.4,Y limit min]
[PIN:D.5,Y limit max]
[PIN:A.11,Z limit min]
[PIN:A.10,Z limit max]
[PIN:A.14,Aux in 0,P0]
[PIN:D.0,Aux in 1,P1]
[PIN:D.2,Aux in 2,P2]
[PIN:A.16,X step]
[PIN:A.3,Y step]
[PIN:C.17,Z step]
[PIN:A.24,X dir]
[PIN:A.2,Y dir]
[PIN:C.15,Z dir]
[PIN:C.6,X enable]
[PIN:A.23,Y enable]
[PIN:B.17,Z enable]
[PIN:C.26,Spindle on]
[PIN:C.24,Spindle direction]
[PIN:C.25,Spindle PWM]
[PIN:C.21,Mist]
[PIN:C.22,Flood]
[PIN:C.14,Aux out 0,P0]
[PIN:C.13,Aux out 1,P1]
[PIN:C.12,Aux out 2,P2]
[PIN:C.29,Aux out 3,P3]
[PIN:D.7,Aux out 4,P4]
[PIN:D.8,Aux out 5,P5]
[PIN:A.8,RX,Primary UART]
[PIN:A.9,TX,Primary UART]

This? Right?

Deltaplc, 2024-11-14:
[VER:1.1f.20241113:]
[OPT:VNM0SL,100,1024,3,0]
Target buffer size found
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,SED]
[FIRMWARE:grblHAL]
[SIGNALS:RHSP]
[NVS STORAGE:*FLASH]
[DRIVER:SAM3X8E]
[DRIVER VERSION:240928]
[BOARD:Ramps 1.4]
[AUX IO:3,6,0,0]
[PLUGIN:Feed override v0.01]
[PLUGIN:Events plugin v0.02]

terjeio, 2024-11-14:
This is strange, I just added the plugin to the Due driver and it works for me.
I see I have not added a reboot/hard reset message to the settings description, it is necessary to do.
Can you post your board map file?

Deltaplc, 2024-11-14:
> Puoi postare il file della mappa della tua scheda?

/*
  ramps_1.6_map.h - driver code for Atmel SAM3X8E ARM processor, pin mappings compatible with Ramps 1.6 board

  NOTE: board must be modified for 3.3V IO before use!

  Part of grblHAL

  Copyright (c) 2019-2023 Terje Io

  Mappings according to Re-ARM for NXP LPC1768

  grblHAL is free software: you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation, either version 3 of the License, or
  (at your option) any later version.

  grblHAL is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE. See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License
  grblHAL with Grbl. If not, see <http://www.gnu.org/licenses/>.
*/

#if (N_AUTO_SQUARED && N_AUTO_SQUARED < N_ABC_MOTORS) || N_ABC_MOTORS > 2
#error "Axis configuration is not supported!"
#endif

#define BOARD_NAME "Ramps 1.4"
//#define BOARD_URL "https://github.com/bigtreetech/ramps-1.6"

 // Define step pulse output pins.
#define X_STEP_PORT         PIOA
#define X_STEP_PIN          16  // Due Analog Pin 0
#define Y_STEP_PORT         PIOA
#define Y_STEP_PIN          3   // Due Analog Pin 6
#define Z_STEP_PORT         PIOC
#define Z_STEP_PIN          17  // Due Digital Pin 46

// Define step direction output pins.
#define X_DIRECTION_PORT    PIOA
#define X_DIRECTION_PIN     24  // Due Analog Pin 1
#define Y_DIRECTION_PORT    PIOA
#define Y_DIRECTION_PIN     2   // Due Analog Pin 7
#define Z_DIRECTION_PORT    PIOC
#define Z_DIRECTION_PIN     15  // Due Digital Pin 48

// Define stepper driver enable/disable output pin(s).
#define X_ENABLE_PORT       PIOC
#define X_ENABLE_PIN        6   // Due Digital Pin 38
#define Y_ENABLE_PORT       PIOA
#define Y_ENABLE_PIN        23  // Due Analog Pin 2
#define Z_ENABLE_PORT       PIOB
#define Z_ENABLE_PIN        17  // Due Analog Pin 8

// Define homing/hard limit switch input pins.
#define X_LIMIT_PORT        PIOC
#define X_LIMIT_PIN         28  // Due Digital Pin 3
#define Y_LIMIT_PORT        PIOD
#define Y_LIMIT_PIN         4   // Due Digital Pin 14
#define Z_LIMIT_PORT        PIOA
#define Z_LIMIT_PIN         11  // Due Digital Pin 18

// Define homing/hard limit switch input pins.
#if X_AUTO_SQUARE
#define M3_LIMIT_PORT       PIOB
#define M3_LIMIT_PIN        25  // Due Digital Pin 2
#else
#define X_LIMIT_PORT_MAX    PIOB
#define X_LIMIT_PIN_MAX     25  // Due Digital Pin 2
#endif
#if Y_AUTO_SQUARE
#define M3_LIMIT_PORT       PIOD
#define M3_LIMIT_PIN        5   // Due Digital Pin 15
#else
#define Y_LIMIT_PORT_MAX    PIOD
#define Y_LIMIT_PIN_MAX     5   // Due Digital Pin 15
#endif
#if Z_AUTO_SQUARE
#define M3_LIMIT_PORT       PIOA
#define M3_LIMIT_PIN        10  // Due Digital Pin 19
#else
#define Z_LIMIT_PORT_MAX    PIOA
#define Z_LIMIT_PIN_MAX     10  // Due Digital Pin 19
#endif

// Define ganged axis or A axis step pulse and step direction output pins.
#if N_ABC_MOTORS > 0
#define M3_AVAILABLE
#define M3_STEP_PORT        PIOD
#define M3_STEP_PIN         1   // Due Digital Pin 26
#define M3_DIRECTION_PORT   PIOD
#define M3_DIRECTION_PIN    3   // Due Digital Pin 28
#define M3_ENABLE_PORT      PIOA
#define M3_ENABLE_PIN       15  // Due Digital Pin 24
#endif

// Define ganged axis or B axis step pulse and step direction output pins.
#if N_ABC_MOTORS == 2
#define M4_AVAILABLE
#define M4_STEP_PORT        PIOC
#define M4_STEP_PIN         4   // Due Digital Pin 36
#define M4_DIRECTION_PORT   PIOC
#define M4_DIRECTION_PIN    2   // Due Digital Pin 34
#define M4_ENABLE_PORT      PIOD
#define M4_ENABLE_PIN       9   // Due Digital Pin 30
#endif

// Define driver spindle pins

#if DRIVER_SPINDLE_PWM_ENABLE
#define SPINDLE_PWM_TIMER       (TC2->TC_CHANNEL[0])
#define SPINDLE_PWM_CCREG       2
#define SPINDLE_PWM_PORT        PIOC
#define SPINDLE_PWM_PIN         25  // Due Digital Pin 5  // M3
#endif

#if DRIVER_SPINDLE_DIR_ENABLE
#define SPINDLE_DIRECTION_PORT  PIOC
#define SPINDLE_DIRECTION_PIN   24  // Due Digital Pin 6  //M4
#endif

#if DRIVER_SPINDLE_ENABLE
#define SPINDLE_ENABLE_PORT     PIOC
#define SPINDLE_ENABLE_PIN      26  // Due Digital Pin 4
#endif

// Define flood and mist coolant enable output pins.
#define COOLANT_FLOOD_PORT  PIOC
#define COOLANT_FLOOD_PIN   22  // Due Digital Pin 8  M7
#define COOLANT_MIST_PORT   PIOC
#define COOLANT_MIST_PIN    21  // Due Digital Pin 9  M8

// Define user-control CONTROLs (cycle start, reset, feed hold) input pins.
#define RESET_PORT          PIOA
#define RESET_PIN           22  // DUE Analog Pin 3
#define FEED_HOLD_PORT      PIOA
#define FEED_HOLD_PIN       4   // DUE Analog Pin 4
#define CYCLE_START_PORT    PIOA
#define CYCLE_START_PIN     6   // DUE Analog Pin 5

#define AUXINPUT0_PORT      PIOA
#define AUXINPUT0_PIN       14
#define AUXINPUT1_PORT      PIOD
#define AUXINPUT1_PIN       0
#define AUXINPUT2_PORT      PIOD
#define AUXINPUT2_PIN       2
#define AUXINPUT3_PORT      PIOB  // Due Digital Pin 21
#define AUXINPUT3_PIN       13

#if PROBE_ENABLE
#define PROBE_PORT          AUXINPUT3_PORT
#define PROBE_PIN           AUXINPUT3_PIN
#endif

#define AUXOUTPUT0_PORT     PIOC
#define AUXOUTPUT0_PIN      14     // Pin 49
#define AUXOUTPUT1_PORT     PIOC
#define AUXOUTPUT1_PIN      13     // Pin 50
#define AUXOUTPUT2_PORT     PIOC
#define AUXOUTPUT2_PIN      12     // Pin 51

#define AUXOUTPUT3_PORT     PIOC
#define AUXOUTPUT3_PIN      29  // Due Digital Pin 10
#define AUXOUTPUT4_PORT     PIOD  
#define AUXOUTPUT4_PIN      7     // Due Digital Pin 11 
#define AUXOUTPUT5_PORT     PIOD
#define AUXOUTPUT5_PIN      8     // Due Digital Pin 12 
/**/

Deltaplc, 2024-11-14:
I copied the contents of the ramps_1.6_map.h file into my_machine_map.h (which I use at the moment) and redefined some pin outputs.... So the ramps_1.6_map.h file remained unchanged.

terjeio, 2024-11-14:
Your map file works for me, after erasing flash, flashing the binary and resetting plugin settings with `$RST=&` and then configuring:

```
$750=3
$751=4
$752=0
$753=0
$760=0
$761=1
$762=2
$763=3
```
I have a genuine Arduino Due.

Deltaplc, 2024-11-15:
> Your map file works for me, after erasing flash, flashing the binary and resetting plugin settings with `$RST=&` and then configuring:

Good morning
Today I tried...
I did $RST=&
I set:
$750=3
$751=4
$752=0
$753=0
$760=0
$761=1
$762=2
$763=3

This seems to be working now!
M7 - M8 bind to their respective outputs.
I also tested Spindle enable (M3/M4), Laser enable (M3/M4):

I did $RST=& (If I don't do this sometimes it seems to not save the parameters correctly!)
I set:
$750=1
$751=2
$752=3
$753=4
$760=0
$761=1
$762=2
$763=3

Everything works except "Spindle enable (M3/M4)" Doesn't activate any output?
The other aux output pairings work correctly.
Conclusion:
1. I turned off the power and restarted the Arduino and the aux ports started working....
2. I did $RST=& , which probably helps to reset the previous settings.
3. Tested the various combinations with events
4. "Spindle enable (M3/M4)" Doesn't activate any output?

SPINDLE_ENABLE_PIN 26 // Two Digital Pin 4pin, in my case it is always set high except if M3 S (Greater than zero),
so the paired output should work the same way? Am I wrong?

Thank you for your support!

terjeio, 2024-11-15:
> I did $RST=& (If I don't do this sometimes it seems to not save the parameters correctly!)

Flashing the DUE clears all setting bytes to 0xFF, and this sometimes causes issues because the firmware is not always able to detect this and automatically reset to defaults. Grbl and grblHAL both uses an 8-bit checksum for detection, I should switch to 16-bit to improve it...

> "Spindle enable (M3/M4)" Doesn't activate any output?

Spindle enable does not work when $32=1 (laser mode enabled), Laser enable should but does not due to a bug. I have attached a patched version that does and also adds binding information to the $pins report.

> SPINDLE_ENABLE_PIN 26 // Two Digital Pin 4pin, in my case it is always set high except if M3 S (Greater than zero),
so the paired output should work the same way? Am I wrong?

Signal inversion is done in the driver and this is not reflected in the aux outputs. You can invert the aux outputs separately with $372.

Also be aware the spindle enable signa isl sometimes beeing controlled by setting spindle speed. Laser mode does that. There are two separate calls for this and I only trap the main one for now for setting the aux output. So e.g. in laser mode the aux output will stay enabled for as long as M3/M4 is active regardless of the spindle enable pin state. This so the aux output can be used for controlling a fan or a water pump without constantly switching on and off...

[eventout.zip](https://github.com/user-attachments/files/17770715/eventout.zip)

Deltaplc, 2024-11-15:
OK, thank you so much for your support!!

I have a question that doesn't focus on aux outputs....

I made the transition from ArduinoMega to ArduinoDue with ramps.
With Arduino Due and ramps when I press the reset button the D10 output on the ArduinoDue linked to pin 10 is activated. It is also activated when I insert the USB cable...
The LED linked to pin 13 (theoretically linked to the reset) never lights up!
With Arduino Mega and ramps everything works well (same ramps). It almost seems to behave as the LED connected to pin 13 of a classic Arduino should behave....
Can you explain to me why?

Deltaplc, 2024-11-15:
> D13 non è collegato a nessun circuito di reset
I explained myself badly...

> Forse. Hai Aux 3 collegato a D10, quindi questo potrebbe spiegare l'attività su quello

I don't think it's for this reason, I also tried to define Pin 10 (D10) with another Aux output and also not to define it....
The problem remains!
In fact, now while testing and loading the firmware on the Arduino Due, I noticed that D10 remained active for the entire loading time!!
I tried with two ArduinoDue and two different ramps, and the problem is on both...
Are you saying the problem lies in the bootloader? The ArduinoDue are not original...they are clones.(aliexpress)
How can I check and solve this annoying problem?

Deltaplc, 2024-11-16:
> Maybe. You have Aux 3 connected to D10 so that may explain activity on that. Connecting USB causes a reset so should behave the same as pressing the reset button. D13 is not linked to any reset circutry so will not lit up unless the bootloader (or grblHAL) does it from code - and IMO it should not.

Good morning
A little update on the D10 issue....
I did some tests, and it seems that in my ArduinoDue clones (without ramps), when you press the reset button or the serial port is activated, all the pins take on a value of about 3.2V (you can say high)!!!
In conclusion I think it is a problem linked to the fact that they are ArduinoDue clones and that they have a bug!!!
I will try to buy an original ArduinoDue and see if the behavior will be different!

I thank you infinitely for the support and time you dedicated to me!!

terjeio, 2024-11-16:
> I will try to buy an original ArduinoDue and see if the behavior will be different!

Mine does not pull D10 high on a reset, but with no load it seems to do so - due to the pin beeing configured as input with pullup enabled. A 68K load brings it down to a little less than 2V. Ensure that any circuitry driven by D10 loads the pin enough to bring it down to a level below what is needed to switch the final output.

Deltaplc, 2024-11-16:
> but with no load it seems to do so

So can you confirm that the original ArduinoDue also seems to pull all pins high when reset is pressed?
Don't you need to buy another original one?

> due to the pin beeing configured as input with pullup enabled.

Is there a method to change this type of setting and solve the problem?
Thank you in advance for your help

terjeio, 2024-11-16:
> So can you confirm that the original ArduinoDue also seems to pull all pins high when reset is pressed(Without ramps)?

It is soft pulling via a weak pullup resistor, not hard pulling via the output stage.
And this might be random? Table 31-3 in the PRM does not specify a reset value for pullup and pulldown enable.
You may have to study the PRM in detail and/or contact Atmel to get a definitive answer?

FYI when checking a STM32 PRM it specifies some pins as floating and some with pullup enabled on reset. So at least the behaviour is documented.

> Don't you need to buy another original one?

I do not need to buy a new one since I only use the one I have for bench testing. And I can design around it if needed. I had to do that when I made a breakout board for a FPGA device.

> Is there a method to change this type of setting and solve the problem?

Load the pin - e.g. with a 10K resistor to avoid it. AFAIK you cannot change the reset state by configuring the MCU.

Deltaplc, 2024-11-17:
Good morning

> Non ho bisogno di comprarne uno nuovo, perché uso solo quello che ho per i test al banco. E posso progettarlo se necessario. Ho dovuto farlo quando ho realizzato una scheda breakout per un dispositivo FPGA.

Sorry I expressed myself badly....I use Google Translate...
I meant that if the original Arduino Due also has the same behavior, it makes no sense to me, buying one to replace my Arduino Due clones!

> potresti dover studiare il PRM in dettaglio e/o contattare Atmel per ottenere una risposta definitiva?

I'll try contacting the help center!

I congratulate you on your GrblHal, it has many potentials and features that are useful! And you will surely add more! For the future, even if it is not a fundamental thing, together with the LEDs (which can also have a signaling function for safety on/off processing, etc...) a buzzer that signals the start and end of processing can be useful! :)

As always, thank you for your support!
Thank you!!
Greetings

closed 2025-04-07 by Deltaplc