grblHAL/SAM3X8E #30: Possibility to impose a speed different from the maximum speed of the translation movement in G0 movements

Deltaplc, 2024-11-05:
Good morning
I apologize for the inconvenience
I use ligtburn to drive a laser where I installed your grblhal.
To move (for example) from the starting point to the engraving area, use the G0 command,
given that in my case I use a very high max travel speed,
I wanted to know if it was possible to set a value for the movements in G0 mode other than the max travel value of the axes?
Maybe even default to the entire firmware.... or it would be great if you could add a $ setting to do this!
Thanks for the help

terjeio, 2024-11-11:
You can reduce G0 speed to 50% or 25% of max by sending [speed override commands](https://github.com/grblHAL/core/blob/1444c08997827a5653eea821f4739fdc8a689a78/grbl.h#L118-L120). The sender you use may have UI controls for doing that.
A user plugin could implement a M-code to dynamically set the G0 override to any percentage by calling [plan_feed_override()](https://github.com/grblHAL/core/blob/1444c08997827a5653eea821f4739fdc8a689a78/planner.c#L665), IMO better than having a new core setting for a default percentage. But of course a plugin could add a setting in addition to a M-code...

Deltaplc, 2024-11-11:
> You can reduce G0 speed to 50% or 25% of max by sending [speed override commands](https://github.com/grblHAL/core/blob/1444c08997827a5653eea821f4739fdc8a689a78/grbl.h#L118-L120). The sender you use [...]

Thanks for the reply
Sorry but I don't understand....with which command can I set the G0 speed override?
If I change "Override speed G0", does the maximum speed of the G1 commands also change accordingly or does it remain unchanged?

terjeio, 2024-11-11:
> Sorry but I don't understand....with which command can I set the G0 speed override?

There are three [single character real-time commands](https://github.com/grblHAL/core/blob/1444c08997827a5653eea821f4739fdc8a689a78/grbl.h#L118-L120) that are not easy to send manually.

Some senders, like ioSender, has override buttons that sends the commands:
[image]
Other senders may have different UI element(s) for doing the same, and some not allowing to change the speed.

> If I change "Override speed G0", does the maximum speed of the G1 commands also change accordingly or does it remain unchanged?

It remains unchanged.

Deltaplc, 2024-11-11:
The basic problem is that lightburn uses the G0 command by default, for all movements that do not concern machining.
I attach a small gcode:

1. G00 G17 G40 G21 G54
2. G90
3. M4
4. ; Cut @ 920 mm/min, 54% power
5. M8
6. G0 X41Y67
7. M3
8. M3
9. G1 F100 S400
10. G4 P0.01
11. G1 S0
12. G0
13. M3
14. ; Layer C00
15. G1 X68S540F920
16. G1 Y37
17. G1 X41
18. G1 Y67
19. G0
20. G4 P0.01
21. M9
22. G1 S0
23. M5
24. G90
25. ; return to user-defined finish pos
26. G0 X0Y0
27. M2
As you can see, use the G0 command to approach the processing point (line 6) and to return to the starting point (line 26).
My laser has no mechanical problems to go at this speed....but I just think it's useless to do it at maximum speed! For this reason I thought that by separating the max speed of the G1-G2-etc. commands. from speed G0 was the best solution, to give more flexibility to the settings...
I don't know, maybe I'm wrong? What do you think?

Deltaplc, 2024-11-11:
> > Se modifico "Override speed G0", cambia di conseguenza anche la velocità massima dei comandi G1 oppure rimane invariata?
> 
> Rimane invariato.

Ok, this might solve my problem if I could add a gcode command to startup.....

1. ;USER START SCRIPT
2. m7
3. m8
4. 
5. G4 P1
6. 
7. G4 P1
8. 
9. new speed override commands?

Can I do it? What command should I enter?

terjeio, 2024-11-11:
A user plugin can be made that implements a M-code, closest I have found by a quick search is [M220](https://marlinfw.org/docs/gcode/M220.html), but it is for all motion. A extra parameter could be added to apply it only to rapids? Or perhaps there is a industry stadard M-code that is used by some CNC manufacturers?

Deltaplc, 2024-11-11:
> O forse esiste un codice M standard del settore utilizzato da alcuni produttori CNC?

No, I don't know of any cnc codes that only set the speed of G0 movements.
Because (unfortunately) it is always linked to the maximum speed of all other movements!

Deltaplc, 2024-11-11:
> Si potrebbe aggiungere un parametro extra per applicarlo solo alle rapide?

Yes 
I think that would be the best thing! It would make everything much more flexible!

terjeio, 2024-11-13:
I have now [added a plugin](https://github.com/grblHAL/Plugins_misc/tree/main?tab=readme-ov-file#feed-override), but not yet added it to all the drivers and the Web Builder. If you are compiling yourself you may add it manually, but remember to update to the latest core first.

Deltaplc, 2024-11-13:
> Ora ho [aggiunto un plugin](https://github.com/grblHAL/Plugins_misc/tree/main?tab=readme-ov-file#feed-override) , ma non l'ho ancora aggiunto a tutti i driver e al Web Builder. Se stai compilando da solo, puoi aggiungerlo manualmente, ma ricorda di aggiornare prima all'ultimo core.

Bench tested plugin!! Seems to work very well!! And exactly what I needed! Actually, better!
Thank you very much!

closed 2024-12-11 by terjeio