grblHAL/SAM3X8E #32: M7-M8-M4 They do not work with the latest core version download

Deltaplc, 2025-03-02:
Hi sorry to bother you...
I'm using this version and everything is fine...

[VER:1.1f.20241113:]
[OPT:VNM0SL,100,1024,3,0]
Target buffer size found
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,SED]
[FIRMWARE:grblHAL]
[SIGNALS:RHSP]
[NVS STORAGE:*FLASH]
[DRIVER:SAM3X8E]
[DRIVER VERSION:240928]
[BOARD:Ramps 1.4 Davide]
[AUX IO:3,6,0,0]
[CLUSTER:16]
Cluster size found
[PLUGIN:LightBurn clusters v0.06]
[PLUGIN:Feed override v0.01]
[PLUGIN:Events plugin v0.03]

I downloaded what I think is the latest version....

[VER:1.1f.20250228:]
[OPT:VNMSL,100,1024,3,0]
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,SED]
[FIRMWARE:grblHAL]
[SIGNALS:RHSP]
[NVS STORAGE:*FLASH]
[DRIVER:SAM3X8E]
[DRIVER VERSION:250228]
[BOARD:Ramps 1.4_Davide]
[AUX IO:2,6,0,0]
[CLUSTER:16]
[PLUGIN:LightBurn clusters v0.06]
[PLUGIN:Laser coolant v0.06]
[PLUGIN:Feed override v0.01]
[PLUGIN:Events plugin v0.05]

With this version the outputs of the m7 m8 m3 commands do not activate! And also the output on the enable pin does not work!
I am attaching a part of my pin map.....

#if DRIVER_SPINDLE_DIR_ENABLE
#define SPINDLE_DIRECTION_PORT  PIOC
#define SPINDLE_DIRECTION_PIN   24  // Due Digital Pin 6  //M4
#endif
#if DRIVER_SPINDLE_ENABLE
#define SPINDLE_ENABLE_PORT     PIOC
#define SPINDLE_ENABLE_PIN      26  // Due Digital Pin 4
#endif
// Define flood and mist coolant enable output pins.
#define COOLANT_FLOOD_PORT  PIOC
#define COOLANT_FLOOD_PIN   22  // Due Digital Pin 8  M7
#define COOLANT_MIST_PORT   PIOC
#define COOLANT_MIST_PIN    21  // Due Digital Pin 9  M8

Deltaplc, 2025-03-02:
Have you added any configuration to enable these features? I can't figure it out!?

Deltaplc, 2025-03-02:
I wanted to update to the new version because I saw that you added the jerk control....
But unfortunately I found myself with the laser blocked!!!

terjeio, 2025-03-02:
Board map format has changed so you will have to [update your map file accordingly](https://github.com/grblHAL/SAM3X8E/blob/84af60193c8c01f187ea1cd6cd50eb89a4a73ab4/grblHAL_Due/src/boards/generic_map.h#L77-L100).

And I see now that the old maps are still present on github - the current ones are in the _boards_ directory. I'll delete old ones.

Deltaplc, 2025-03-02:
OK now I'll try to look and try to replace the pin map!!
Thank you very much!

Deltaplc, 2025-03-02:
Hello
Just to confirm that by redoing the pin map with the new syntax, everything works!
Thank you for your availability!!

I ask you another information, it concerns the jerk management...
I enabled the jerk management (#define ENABLE_ACCELERATION_PROFILES On #define ENABLE_JERK_ACCELERATION On)
But to get the correct speed of the axes, I had to set M800=90000 !! (I'm attaching a couple of parameters $110=25000.000 , $120=4000.000)

Am I doing something wrong? Do I have to enable something else to make the jerk management work?
Maybe it's better to open another discussion?

terjeio, 2025-03-06:
> Maybe it's better to open another discussion?

You can join [this discussion](https://github.com/grblHAL/core/discussions/384).

Jogging is a known issue - as I understand it there is work ongoing to try to fix it.

closed 2025-03-06 by terjeio