grblHAL/SAM3X8E #33: RGB LED strip control  error:20!!

Deltaplc, 2025-04-07:
Hi
Sorry if it bothers you again...
I wanted to activate a WS2812 LED strip, if I understood correctly with : define RGB_LED_ENABLE 2 
I should be able to do it??
But unfortunately I get this error:
error:20
Comando g-code non supportato o non valido trovato nel blocco.

I'm attaching an $I, Does it seem to have loaded correctly?!!
Do I need to do something else? 
Do I need to define the Output Pin? 
I tried but I still get this error??
I don't understand, there's something I'm missing....
ok
[VER:1.1f.20250405:]
[OPT:VNM0SL,100,1024,3,0]
Target buffer size found
[AXS:3:XYZ]
[NEWOPT:ENUMS,RT+,HOME,EXPR,SED]
[FIRMWARE:grblHAL]
[SIGNALS:RHSP]
[NVS STORAGE:*FLASH]
[DRIVER:SAM3X8E]
[DRIVER VERSION:250403]
[BOARD:Ramps 1.4_Davide]
[AUX IO:3,3,0,0]
[CLUSTER:16]
Cluster size found
[PLUGIN:LightBurn clusters v0.06]
[PLUGIN:CO2 laser overdrive v0.01]
[PLUGIN:RGB LED strips (N/A) v0.05]
[PLUGIN:Feed override v0.01]
ok

Deltaplc, 2025-04-07:
I also tried with :#define RGB_LED_ENABLE 1
But no $536/537 appears....
and it always returns 
[PLUGIN: RGB LED Strips (N/A) v0.05]

terjeio, 2025-04-07:
My bad, the SAM3X8E driver does not yet have the lowlevel driver support required. I have now removed the option in _my_machine.h_, added by a mindless copy-paste.

Deltaplc, 2025-04-07:
Ok thanks...
But we need to add a specific pin, with what syntax is it defined?
Will you make it work for SAM3X8E in the future?
With which core can it be enabled?
The Esp32 core, does it support the plugin: RGB LED Strip Control?

terjeio, 2025-04-09:
> But we need to add a specific pin, with what syntax is it defined?

In the map file, something like this:

```
#define LED_PORT GPIOC // rail LED strip
#define LED_PIN  9
```

The driver the has to provide support for the LEDs connected to that pin, preferably via DMA if lights are to be updated during motion (due to strict timing requirements for the LED pulse train). Something similar to [this from the STM32F4xx driver](https://github.com/grblHAL/STM32F4xx/blob/master/Src/neopixel_spi.c), but for the SAM3X8E.

> Will you make it work for SAM3X8E in the future?

Maybe. I would prefer someone else adding the code and create a PR.

> With which core can it be enabled?

ESP32, iMXRT1062, RP2040, STM32F4xx, STM32F7xx and STM32H7xx.

> The Esp32 core, does it support the plugin: RGB LED Strip Control?

Yes. But like the others very few, if any, boards has the pin(s) defined.

Deltaplc, 2025-04-09:
Ok, thanks for the info!
I wanted to try using neopixsel, to make a signal of a machine in operation etc.
But no problem I'll invent some other way!

> Maybe. I would prefer someone else adding the code and create a PR.

Unfortunately I am not a C++ programmer!
I learned C/C++ language by myself! (I do PLC programming).
If I could, I would definitely help you!

Thanks again for your support!